matthuszagh/fmcw-radar #7: Patch antenna

AlexBdx, 2019-07-01:
Hello,

Is this patch antenna the one that Henrik used or your own design based on his post?

matthuszagh, 2019-07-02:
It's basically Henrik's. He gave me the gerbers for an array of 2 but I turned it into a single patch and imported it into kicad (the one patch should be identical to his though). If you're interested in the original gerbers I can give those to you. I'll be turning this into an 8-array antenna (multiplexed through 2 rec channels) and running sims in openems, just fyi if you're interested. Not sure the eta on that

AlexBdx, 2019-07-02:
Yeah I saw your multiplexing board. I have ordered a copy of your array of 2 after talking with Henrik. What connectors did you use? I saw a footprint with 9 holes and was wondering of you had a reference for it.

matthuszagh, 2019-07-02:
Hm 9 holes.. Are you talking about the RF via (reference F1 and F2)? That one only has 6 holes though. If you provide me more info or a screenshot or something I can tell you what it is. For sma cables I'm using [these](https://www.amazon.com/gp/product/B072FS4WMK/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). They work fine. you'll need a transmission antenna too. have you sorted that yet? I'm currently using [this one](https://www.amazon.com/gp/product/B07D28MW63/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) planning to probably build my own later. If you want something cheap the one from the link is good. Unfortunately you lose half the transmission energy because the tx/rx polarizations don't match, but as a first pass to get the board working it's good

AlexBdx, 2019-07-02:
Yep I got that Tx. I saw that in your doc and that will be perfect for a first run (I saw a single feed version mentioned on Martin's website but could not find anyone that sold it).
Regarding the RX, I imported your files from the antenna folder and basically sent that straight to PCB manufacturing. In Kicad, everything appears as a via to me. Below is a the file I am using. Let me know if you can see it.
[Patch_2x5.zip](https://github.com/matthuszagh/fmcw-radar/files/3352443/Patch_2x5.zip)

matthuszagh, 2019-07-02:
ah i see why you're confused. that kicad_pcb file is wrong. fortunately the gerbers in the antenna directory look fine. in any event this is the [surface sma i used](https://www.digikey.com/product-detail/en/linx-technologies-inc/CONSMA001-SMD-G/CONSMA001-SMD-G-ND/4496569). That's all you need, plus the sma connector cables. i'm gonna try to track down the right kicad_pcb file but might not get to it right away.

AlexBdx, 2019-07-02:
Damn! I just uploaded all the files in Gerber (through KiCad) and then Saved As a kicad_pcb file. Did I have to make some modifications beforehand? I just got an email saying that the PCB had gone to the fabricator, but maybe I can cancel the order if the board cannot be easily modified. Let me know!
I'll order the SMA connector you mentioned so at least I have these ready.

matthuszagh, 2019-07-02:
Ah sorry about that! Yeah I'd try to cancel if you can, I think there's a good chance it won't work properly. To clarify, the exposed copper looks like a bunch of overlapping circles?

matthuszagh, 2019-07-02:
For reference the correct gerber looks like this
[image]

AlexBdx, 2019-07-03:
Okay, can you send me the correct files you have? Or maybe update the repo? It might be easier to swap the gerber files at the fabricator than outright cancel as my order is grouped with dozens of others on a panel.

matthuszagh, 2019-07-03:
The gerbers in the repo are right but it looks like kicad doesn't know how to convert it to a kicad_pcb file. I just tried the conversion here. does the mfg allow you to just send gerbers? i'd just use those directly

AlexBdx, 2019-07-03:
Wow yes indeed! I tried to upload the gerber zip that you have in the antenna folder and it looks much better. The attached shape is correct right?
[image]

matthuszagh, 2019-07-03:
Looks good!

AlexBdx, 2019-07-11:
Hey,
I put together the antenna and wired everything, but unfortunately I'm not getting any meaningful data yet. The ADL5802 on the board gets really hot and it seems that the LOIN/LOIP pins are shorted, even if I cannot see any soldering issues on the board. Did you have troubles with this chip at all?

matthuszagh, 2019-07-12:
Not really, the only issues I encountered were a few simple shorts that we are all fixable. The trick was tracking them down.

AlexBdx, 2019-07-13:
Mmmh, I'm a little puzzled. I get some data, but it looks like crap. My channel 2 does not seem to be getting anything meaningful.
Do you have a log file you could share so I can run plot_log and see how the results should look like when it works?
I uploaded one of mine here if you are interested/have time to look at it: [https://github.com/AlexBdx/fmcw3/tree/master/pc/python2/run_away](run_away)
The log file is called run_away.log and in the img/ folder you have the plots I generated - they don't look so good. Let me know!

matthuszagh, 2019-07-19:
https://www.dropbox.com/s/luuqggbdlmfhpul/fmcw.log?dl=0

here's a log i took. you can't really see anything on the angular range plots, but the range time plot is ok. i'm hoping the antenna simulations and multiplexed antenna will yield something with better angular resolution and better snr, but not done with that yet.

I took a look at yours and it seems like you're getting repeating data.. not entirely sure what might be causing that. did you check for shorts at the receiver? I had one on one of the lna's that caused bad data. just a thought.

AlexBdx, 2019-07-19:
in read_settings, it raises a ValueError due to missing error.
if header != 'fmcw3;':
        raise ValueError('Missing header')
Did you modify fmcw3.py and discarded the header?

I have fixed "some" issues on my board (found one tricky short on a non-populated yet connected pad) but I'm done 100% done. I hope to find the others soon.

matthuszagh, 2019-07-19:
oh yeah i changed the name. you have to modify the python code to look for fmcw, not fmcw3

AlexBdx, 2019-07-20:
Do you have the corresponding plot_log.py file too? 
I've tried yours & Henrik's legacy file, removing errors one after another until I would be processing the reading the whole log... only to end up with empty arrays.
Maybe you could send me the one you used so I don't have to reinvent the wheel on this one.
Have a great weekend!

matthuszagh, 2019-07-24:
https://www.dropbox.com/s/3eo8sn8id342ecr/plot_log.py?dl=0

let me know if that works

AlexBdx, 2019-07-24:
Stupid mistake on my end: I was trying to read your file with python 2 (what Henrik initially used) while I had to run it with python 3. Your range-time plot is noisy, but I can definitely see a signal on it!! Looks great!
Yes, it did work. A couple of questions:
- I thought that the IF time domain waveforms would be limited to [-1; 1] V but I see that the signal spiked outside these bounds at times. How is that happening? I thought the ADC's largest sampling range was 2V.
- The IF time domain plots have a ton of noise in your signals. Is that normal? How can the noise be reduced? 
- Each channel on each sample of the IF time domain plot seems to have a mean of about -0.25 V. Why isn't it 0 V?

AlexBdx, 2019-07-29:
I forgot to add one of the pictures I had processes to illustrate my previous message. Let me know when you have a second to look at it. Thanks!

[image]

AlexBdx, 2019-07-31:
Wow I finally fixed it. I found a new issue after checking everything again, and now it works. Check that out:

[image]

closed 2019-07-31 by AlexBdx

matthuszagh, 2019-07-31:
glad you figured it out!