matthuszagh/fmcw-radar #8: Multiplexed fmcw

AlexBdx, 2019-07-31:
Time to get real, now that the board is working.
How far did you get on the multiplexing? I saw your Kicad file, did you:
- Make it?
- Write the code FPGA code to support it?
- Python/pc code for it?
I'm interested in pushing a little forward that idea, so let me know!

AlexBdx, 2019-08-30:
Regarding point 3, I've made some progress actually.
Would you mind testing my version of the fmcw3 python code and let me know if it works on your end too?
You would need to install the files I used with `pip install fmcw` and then run the attached main file. Thanks!
[rt_fmcw3.zip](https://github.com/matthuszagh/fmcw-radar/files/3558005/rt_fmcw3.zip)

matthuszagh, 2019-09-03:
I'll try to test this when I can. What's the purpose of this code? Is it to support antenna multiplexing? If so, my testing will be a bit delayed since I don't yet have the receiver antennas and haven't modified the FPGA code to support it.

One of the main goals of this project for me was to learn FPGA dev, so I've been working to get all functionality (except plotting of course) onto the FPGA. All the individual modules work and now I just need to connect them together and test it, so hopefully I'll have a complete solution (including realtime plotting and antenna multiplexing) soon. I'll post back when I do.

AlexBdx, 2019-09-03:
In a nutshell:
- Work at a higher level than Henrik's code by providing functions/classes/modules... and documenting stuff a lot more
- Add some features

Key features:
- Runs Henrik's core physics
- Display IF signals/Angle/Range time in real time. Optionally, connects to a webcam and records video of the scene imaged.
- Saves data to binary & csv (in csv mode you save the integer values of the IF signal)
- Way more robust management of the USB connection, less than 5% of the sweeps are dropped now.
- Heavy reliance on the multiprocessing library to allow regular laptop to perform okay

AlexBdx, 2019-09-03:
I know ~ nothing about FPGA, so I'm glad you are taking a stab at this challenge. Performing the FFT on the FPGA would definitely be more efficient than on a laptop. However, it's a lot harder to write than a few call to Numpy so if you go that route I would be interested to see how you did it.

matthuszagh, 2019-09-03:
Nice job getting all this working with just python code. I'm looking forward to testing it. I actually think your solution makes more sense for most applications than mine. The FPGA code is more efficient, but less flexible when you want to change something and as you mentioned harder to program. I'll be happy to walk you through it when I'm sure it actually works.

AlexBdx, 2019-09-03:
Thanks! Let me know if my script works for you. If you use windows, there might be some linux specific implementations that derail it.
Also I added my personal script to the fmcw library so you can just run YYMMDD_rt_fmcw3.py from there and see what happens. Thanks!

AlexBdx, 2019-09-14:
I've created a slack channel for the fmcw so we stop writing endless issue threads
[https://join.slack.com/t/fmcwworkspace/shared_invite/enQtNzYyNTcwNDI0MTgyLTU4NzEzMDA3NTU5ZjI2YTY3MTI5ZmQ0ZDFkOGNlNGRjMzgxNzE1NWZhYjgxZDIyNDlkM2U3YjRhY2RhMGNlOTg](url)
Feel free to join!