matthuszagh/fmcw-radar #9: Homemade horn antenna

AlexBdx, 2019-09-03:
I remember you previously said you wanted to make your own antenna.
I found this article: https://antennatestlab.com/3dprinting
and made my own horn antennas for the fmcw3. They cost nothing and are fun to build.

AlexBdx, 2019-09-03:
Purely informational. If you do not have a 3d printer available you can email me.

closed 2019-09-03 by AlexBdx

matthuszagh, 2019-09-03:
This is awesome, thank you for sharing!

matthuszagh, 2019-09-03:
I'm curious, did you build small horn antennas for reception too, or just a large one for transmission?

AlexBdx, 2019-09-03:
If I just want the range-time plot, I use two horns like Henrik does (check out his new article if you have not already).
If I want the angle plots, just for the transmission due to the need to have a specific alignment between the antennas.
That being said, you could probably design a very narrow horn, and stack them side by side with a lambda/2 pitch. The horn would be non-ideal, but it might be better than the FR4 patch \o/

matthuszagh, 2019-09-03:
Yeah I'm not in love with the patches. The bandwidth on them is terrible and FR4 is tough to work with for antennas. I'll see if I can make horns small enough for reception and report back if they work any better.

AlexBdx, 2019-09-03:
Sounds good.

matthuszagh, 2019-09-04:
you used the wr187? what did you do for waveguide coax adapters?

matthuszagh, 2019-09-04:
wr159 is a better frequency range but i don't see an stl model for it

AlexBdx, 2019-09-04:
No I made a WR-159

matthuszagh, 2019-09-04:
is the model there and i missed it or you made it yourself?

AlexBdx, 2019-09-04:
I made a CAD myself for the WR-159 and 3d printed it. I used the following drawing: https://www.pasternack.com/images/ProductPDF/PE9860-15.pdf
My 3D printer can only do 6"x6"x6" so I had to split it in 2. I did not have the motivation to split it in 4 to make a 20 dBi version

AlexBdx, 2019-09-04:
Let me see if I can 1) find the file 2) attach it on Github

AlexBdx, 2019-09-04:
> 
> 
> you used the wr187? what did you do for waveguide coax adapters?

2nd part of the answer: I *could* have 3D printed one, but I concluded it would be too risky. So I just bought this: https://www.ebay.com/itm/232688038314

matthuszagh, 2019-09-04:
makes sense, thanks

AlexBdx, 2019-09-04:
[WR-159_15_dBi.zip](https://github.com/matthuszagh/fmcw-radar/files/3572572/WR-159_15_dBi.zip)
Assembly_15_dBi is for reference: there are two parts due to the size constraints of my printer. Support_PLA is a conical section that holds the top and bottom parts aligned while you weld the two parts together (via solvent or glue or else)